necroware/s7-vrm #2: two question about schematic.

youxiaojie, 2023-05-04:
two errors:
1. feed back voltage should be divided from "sense" pin
2. power good signal should be generate from a transistor.  details: serial with a resistor to baser, then emitter connected with gnd leaving float collector as power good signal.

necroware, 2023-05-04:
Second point is true, the resistor was just a shortcut, but a transistor would be a more appropriate way to solve that. The first point I don't understand, would you like to explain it a little bit more?

youxiaojie, 2023-05-08:
do you think of professional CPU powering chip meeting vrm 8.3？from 1.3 to 3.5v. I decide to do it with experimental board with adp3157. 
the chips are:
CS5165/5155(H)/5157H/5159
LTC1753
IRU3004/3007/3011
HIP6004B/6004C
UCC3588/3830
LM2636/2638
TPS5210/5211
RT9223CS/9224CS/9224PS
L6911B/6911C
ADP3153/3154/3155/3157
RC5051/5052/5053/5054/5055/5057/5058
APW7004
SC1153
US3012

necroware, 2023-05-08:
I wanted to use s.t. what I can still buy new and what is easy to solder by hand.

youxiaojie, 2023-05-09:
> I wanted to use s.t. what I can still buy new and what is easy to solder by hand.

which s.t. ? the part number?

necroware, 2023-05-09:
Initially I started to look at LTC1530, which was designed for AMD K6-2 and Pentium II in mind. However, that is not obtainable anymore, so I looked for something similar, like ISL6545, NCP1587, L6726. I ended up with NCP1579, since that was deliverable at that point. Unfortunately id disappeared as well now and the next revision will probably be around ISL6545, which has slightly different reference voltage, but otherwise very similar.

Dunwood84, 2023-06-01:
Is there a file that has the board dimensions / board thickness / hole spacing /hole size on here? i'd like to make a board, but most PCB sites require board dimensions and hole sizes

necroware, 2023-06-01:
> Is there a file that has the board dimensions / board thickness / hole spacing /hole size on here? i'd like to make a board, but most PCB sites require board dimensions and hole size

Your question is not related to this topic. If you have questions please open new issue the next time, otherwise we will lose track if every question is in the same ticket. Regarding your question you probably make it wrong, just take the gerber.zip file from the releases page and upload it as is to a manufacturer of your choice. There is everything included.

gmiranda, 2023-07-11:
> Initially I started to look at LTC1530, which was designed for AMD K6-2 and Pentium II in mind. However, that is not obtainable anymore, so I looked for something similar, like ISL6545, NCP1587, L6726. I ended up with NCP1579, since that was deliverable at that point. Unfortunately it disappeared as well now and the next revision will probably be around ISL6545, which has slightly different reference voltage, but otherwise very similar.

I've assembled a board using the ISL6545 and, indeed, the voltages are different. Have you already considered what resistors values must be changed?

necroware, 2023-07-12:
@gmiranda Yes, I recalculated the values, but I'm currently not at home. I will try to update the documentation later with optional values.

necroware, 2023-07-13:
> I've assembled a board using the ISL6545 and, indeed, the voltages are different. Have you already considered what resistors values must be changed?

Just updated the documentation with the alternative values, which you'd need for ISL6545. Those number are only from calculation, I didn't test them yet, but they should be fine. If you test it, would be nice to have some feedback.

gmiranda, 2023-08-11:
> Just updated the documentation with the alternative values, which you'd need for ISL6545. Those number are only from calculation, I didn't test them yet, but they should be fine. If you test it, [...]

Thanks, I'll be sure to leave feedback! Unfortunately, it'll take some time before the new components arrive.

gmiranda, 2023-10-23:
Hi @necroware 

I've justed tested a new board assembled with an ISL6545 and the new components. Using a resistor as a load and a cheap multimeter + cheap scope (Zeeweii DSO1511g) I can see that most of the values are ok. They seem a bit low, usually by about 0.025V, which could be attributed to either my measuring equipment or the ATX PSU I'm using to power this thing up.

There's one setting that seem to be lower than the others and it's 1.7V: I get 1.627V. I'll see if it behaves the same with a real motherboard.

I'd love to verify the output ripple just as you did in your video but this scope doesn't seem to allow me in either DC or AC mode.

I'll probably skip ahead and test the VRM board with a real motherboard. I was able to boot both an MMX and a K6-2 with the old values so I'm confident it'll work as well.

Is there any particular thing you'd like me to test?

Thanks!

necroware, 2025-07-16:
Outdated. New revisions addressed this topics.