necroware/s7-vrm #3: Question about gerber file generation

gmiranda, 2023-06-16:
Hi, I'm new to PCB printing and this is probably a dump question.

I'm trying to generate the gerber files in order to send this to PCBWay and KiCad is telling me that "Zone fills are out-of-date. Refill?"

I don't know what this means. Is it normal or am I missing a layer? 

These are the options I'm using when plotting:
[image]

Thanks a lot!

necroware, 2023-06-17:
You probably have changed something on the layout or may be used different version of KiCad. There is no need to generate gerber files yourself, I wouldn't even suggest it, since on the main branch there can be an untested "work in progress" state. It's better to use the gerber files from the last official release. You can find it here: https://github.com/necroware/s7-vrm/releases

Just take the [s7-vrm-0.4-gerber.zip](https://github.com/necroware/s7-vrm/releases/download/rev-0.4/s7-vrm-0.4-gerber.zip) file and drop it as is by a manufacturer of your choice.

gmiranda, 2023-06-18:
Oh, I didn't realise it was that easy. Thanks a lot!

closed 2023-06-18 by gmiranda