necroware/s7-vrm #4: Is this inductor compatible?

gmiranda, 2023-07-02:
Hi again,

I ordered from LCSC this inductor PROD Tech PDPTAT0810-3R3M  (code C5120885), one of the few that were available. I did not, however, take into account the current specs... It is listed as having an max RMS current of 5400 and max ISAT current of 7000.

Will that be enough? I plan on running an  MMX 166 (2.8V) and a K6-2/450AFX.

The TDP of the Pentium is, according to Intel, 13.1 W. At 2.8V that would be some 4.68A. The K6, according to [Wikichip](https://en.wikichip.org/wiki/amd/k6-2/k6-2-450afx), is 18.8W, so that's 8.55W at 2.2V...

I only have basic electronic skills. Perhaps you could suggest a couple of valid part numbers in the doc?

Thanks!

necroware, 2023-07-02:
Hi, as I stated in the video and written in the documentation, I didn't find proper coil at LCSC. That's why I simply made it myself. The inductor doesn't need to be very exact, anything between 2.5µH and 4.7µH should work, but the sweet point is at around 3.3µH. The inductor can be self made by using a T50 ferrite toroid. For example T50-26 with permeability 75µ wrapped in 10 loops of 1.3 mm copper. It works also just fine with down to 8 loops and 1mm copper.

gmiranda, 2023-07-02:
Thanks for the reply. Sorry, I didn't recall that comment on the video. I did read the doc but assumed the one I found would just work... Will keep looking.

necroware, 2025-07-16:
Outdated.