necroware/s7-vrm #5: VRM starting problem.

Ghostdog85, 2024-01-18:
Hello, I've build vrm according your instruction. It works well, except one issue with it's starting. It happens time to time, not every boot, it can happen on cold boot or after turn off and on. I have noticed that it happen more often when voltage is set 2.0 or more. It also happen when it's set 1.8 but not so often. It seems  it doesn't happen on 1.6-1.7 volts. I tried different more powerful PSU and tried to remove C8 capacitor but it didn't change anything.

fandjelo, 2024-01-18:
Hi, with the listed parts I didn't encounter any problems, so it's going to be hard to reproduce this ob my side. Which DC-DC controller do you use? The startup sequence is highly dependent from the compensation network C7,C8,R1. It's a kind odt hard to find universal circuit for such a big range of voltages and slightly unbalanced parts can heavily affect the start of the converter. If you have problems with that, I'd also suggest to remove Q1 for experiments. It is used for shutting down, but most mainboards don't use this signal anyway and it can affect the compensation network.

necroware, 2024-01-18:
Hi, with the listed parts I didn't encounter any problems, so it's going to be hard to reproduce this ob my side. Which DC-DC controller do you use? The startup sequence is highly dependent from the compensation network C7,C8,R1. It's a kind odt hard to find universal circuit for such a big range of voltages and slightly unbalanced parts can heavily affect the start of the converter. If you have problems with that, I'd also suggest to remove Q1 for experiments. It is used for shutting down, but most mainboards don't use this signal anyway and it can affect the compensation network.

Ghostdog85, 2024-01-18:
I used NCP1579. I'll try to desolder Q1 and test. Also I used 15,2K R1 resistor instead of 15K. Could it affect the starting of vrm ?

Ghostdog85, 2024-01-20:
Hi, it seems like removing Q1 fixed the issue. Now it turning on every time.  As Q1 I also used DTC143ECA.

necroware, 2024-01-21:
Ok, thank you for your feedback. Then we have to find an alternative part for the Q1, something with much higher internal resistance.

Ghostdog85, 2024-01-28:
I'm not sure that I understand what the disable signal is for.  Is it part of MMX self protection technology ? I'm using this VRM with K6-3+ CPU. Maybe it is not nessasery for AMD processors ?

necroware, 2024-01-28:
It's not neccessary indeed. I just followed the specs in the implementation. Some mainboards have a jumper to switch between different VRM modules and they can be deactivated using that signal. Probably also for delayed startup this can be used, but so far, I didn't see any mainboards, which would really need this. If it doesn't work in your case, just remove it. Please don't forget, though this VRM works, it is still not perfect and needs more testing and improvements in the future. May be we can avoid disable function completely.

necroware, 2025-07-16:
The stability on newer revisions was improved, so this ticket is mostly outdated.