necroware/s7-vrm #6: Confirmed working: IBM Aptiva

Geoman, 2024-08-23:
IBM Aptiva 2168-931
Mainboard OEM: Universal Scientific Industrial Co., Ltd
Chipset: Opti 82C557M Viper-N+
BIOS: BGOGR1L 12/29/98
CPU: AMD-K6-III+, 400 Mhz, 66 MHz Bus Clock, Multiplier 6x
Socket 7
4x32MB Kingston EDO RAM 60ns
s7-vrm revision 0.4 running flawlessly
prior I had a Kingston PowerChip-Socket for split-voltage in use, that ran extremely hot, sometimes the PC did not start

Geoman, 2024-08-23:
K6-III+/400ATZ, no modding necessary

necroware, 2024-08-23:
Wow! Very cool! An Opti board this is a great news, that it works there as well. Thank you very much. I will add it to the list of tested boards then. Is it this one? https://theretroweb.com/motherboards/s/universal-scient-pro-57a

Geoman, 2024-08-29:
> Wow! Very cool! An Opti board this is a great news, that it works there as well. Thank you very much. I will add it to the list of tested boards then. Is it this one? https://theretroweb.com/motherboards/s/universal-scient-pro-57a

Yes, this is it exactly!