necroware/s7-vrm #8: Revision 0.7 sense pin

Gouster4, 2025-01-05:
Using sense pin was awesome idea, and voltage can be more precise.
I understand that it can lead into issues on some boards, thats why version 0.7 came out.
Thats why i thougt that instead of completly giving up on sense pin, it would be ideal to be able to switch between using sense pin or using regulator output for sensing.

First i thought about adding one more row to dip swtich, but those are just two pin switeches that connect or disconnects two contacts from each other. In our case we would need 3 pin switch to switch sense between regulator output or sense pin. Easiest way would be 3 pins and jumper. But we dont need carry lot of current here, and added potential resistance of oxidated contacts (they can oxidate over time, specially when scraped while pluggin in and out).

So, solution that i thought about at end was small 3pin switch that can be seen on many laptops as switch that turns on/off wlan.
I attached images of some possibilities for switch that came to my mind.
[image]
[image]
[image]

necroware, 2025-01-05:
I thought about that, but the sense pin is there to compensate the connector resistance and losses on the mainboard. That resistance is quite low, so the sense pin would compensate just a little bit. Adding another switch or jumper would introduce additional resistance, which would push the error in the other direction. So, without the sense pin at all, we would get slightly lower resistance. With a switch/jumper we would potentially get slightly higher voltage, dependent on the quality and purity of the switch/jumper. I also thought about doing a zero voltage resistor, or solder jumper, but then it requires soldering skills to switch. So, eventually I decided to drop it completely. As I described in the release, it's more important to maintain compatibility with as many boards as possible, than to have a super exact voltage. Ignoring sense pin would still give us a very good voltage approximation.

Gouster4, 2025-01-05:
> I thought about that, but the sense pin is there to compensate the connector resistance and losses on the mainboard. That resistance is quite low, so the sense pin would compensate just a little [...]

I understand. So, switch or any aditional mechanical part provides worse sensing than one used from regulator output.

Then i guess, solder jumper can be the way, while between solder jumper pads can be thin via connecting regulator sensing to regulator output as rev7 was meaned, but it would be possible to easely scrape that via and solder bridge between sense pin pad and regulator sensing pad in case someone would want to use it.

So basically solder jumper would be bridged by default to wider support instead of using sense pin, but via this tiny via between pads instead of solder blob.

Also, average between voltage on regulator output and one that comes from sense pin is about precise voltage on cpu. because both vias on motherboard are leading from same vrm to same cpu, just different pin numbers. that makes it like two almost identical resistors, and voltage between two same resistors is exactly (V1+V2)/2, so mathematical average between those two.
But i dont know if such a averaging voltage can be even implemented to voltage sensing of regulator. It was just my thought how i think it works, but i dont have any clue of implementing it. Maybe even if it would be possible, it would be too much effort with miserable benefit.

necroware, 2025-01-05:
That's all is as always a question of effort. A soldered wire from the sense pin to vcore is also an option. A real jumper or switch is definitely not great, because it will add additional resistance as I explained. I think for now it's not terrible to ignore the sense pin completely, but may be in the future revisions I will add something. Let's see what kind of good ideas the future brings.

Gouster4, 2025-01-06:
[image]
I imagined something like this.
Via is on PCB that does connect vrm sensing to vrm output.
But red drawing shows how via can be cut, and other two pins bridges to get sense pin working again in case that user would be sure that his board does support it.

closed 2025-01-06 by Gouster4