necroware/s7-vrm #9: Planning on installing on IBM PC 330

priangle, 2025-03-13:
Hey Necroware! Happy to see such a project. My machine is [IBM PC 330 (Model 6577)](https://theretroweb.com/motherboards/s/ibm-pc-330-350-series-type-6577,-6587) and i am planning on making your VRM module for it. I am a bit worried for the compatibility though, because the machine is OEM, and who know what proprietary modifications to the vrm module IBM has made.
I found the [schematics for the original VRM module for this exact machine](https://www.digchip.com/datasheets/download_datasheet.php?id=675559&part-number=MP55C-3.3-2.8) and the pinout scheme seems to be identical to that you showed in your [first youtube video](https://www.youtube.com/watch?v=CMiGVQbMC5U) (9:30).

[image]
this is how the motherboard socket look right now

[image]
i've got myself a cheap donor motherboard (to the right) to solder out the proper VRM socket, gonna begin the process by soldering it to the IBM board. Only there's some resistors in the way, in the VRM socket area, they are about a half a mm in height, and they prevent the socket to lay flat on the board and the pins to puncture it properly. Not sure what to do about them, except drilling the corresponding areas of the socket on the router for it to have the space for those resistors :/

necroware, 2025-03-13:
Looks interesting. Meanwhile I tested some boards with my adapter and it turned out that not all of those follow the specs. You have to check couple of things. 

I see, that you already have the latest rev 0.7, which works around an issue with the SENSE pin which is not connected to VCORE on some boards. So if your board has such an issue, that wouldn't be a problem with this revision of the module.

Following steps you will have to do without (!) a CPU installed.

You will have to remove resistors R46, R47, R52, R53. Those are the same as the jumper wires which I removed in my first video. Those 0 Ohm resistors short 3.3V I/O voltage to VCORE. You don't want this, since S7-VRM shall control the voltage. 

Regarding R27, R38, R15, R18 you will have to find out what those resistors are  connecting using a multimeter in continuity mode. I assume those resistors should be also removed in case a VRM module is used, but you will have to analyze those resistors aim. Try to make a schema and send a picture here.

After you are done with resistors, before you install the VRM header check out that the I/O voltage pins are NOT connected to VCORE. Unfortunately I found one board where those pins were permanently connected. The board had no I/O and VCORE plane split and despite having a VRM header it was not possible to install any. That was definitely a hardware design bug which I hope your board doesn't have.

Don't hesitate to answer your questions here, I'd be glad to add to the documentation if this board is compatible or not.

priangle, 2025-03-13:
Thanks you so much for a quick and thorough reply! Going to do everything you mentioned. Regarding the resistors, maybe that explains why they obstruct the proper placement of a plastic vrm header - because they need to be removed when using the vrm module.  

forgot to mention that the datasheet for the vrm module has been found on [this vogons thread](https://www.vogons.org/viewtopic.php?t=81718), where OP has the same system as me.

priangle, 2025-03-13:
another hint i found is that the official IBM module (P/N 76H3658) is called ["IBM Aptiva Voltage Regulator"](https://www.memory4less.com/ibm-cpu-vrm-and-accessory-76h3658) on memory4less.com, which is interesting (if it's not a naming error) because another user here already confirmed that your module [works on Aptiva](https://github.com/necroware/s7-vrm/issues/6)

necroware, 2025-03-13:
Ok, that looks promising.

priangle, 2025-03-14:
in order to analyze resistors R38, R27, R18, R15 i've removed them first, then checked both of their solder points separately for all 30 vrm pins

this is how it looks right now:
[image]

so using a multimeter i've made this map, in which:
green connection means no resistance between two points
yellow connection means varied resistance (value changes when swapping red and black needles of multimeter)
red connection means no connection in one direction, and a lot of resistance the other way
no connection means no connection
[image]

as you can see, R38 and R27 are identical, as well as R18 and R15

hope the scheme is readable! :)

necroware, 2025-03-14:
Wow, that is a very nicely made scheme. R18 and R15 look to be the same as R46, R47, R52, R53 - those resistors are there to bridge 3.3V onto VCORE. Unfortunately looking at this it doesn't look good since on all of your schemas you seem always to have I/O connected with VCORE. It could be that this board has the same issue as what I mentioned above. Looks like the I/O plane is not separated from the VCORE plane. If you remove all resistors, you should not have connection between the VRM pins A4, B3, B4 and A7, A8, A10, B7, B8, B9, B10.

If those pins are connected, than you can't control I/O and VCORE voltage separately. Are the more 0 Ohm resistor around which are connected to the named pins?

priangle, 2025-03-14:
here are the closest 0Ohm resistors
[image]

necroware, 2025-03-14:
I saw those, they are actually too small, for power transmission. On the other hand I/O voltage doesn't need too much. Are some of those resistors connected to the I/O voltage pins?

priangle, 2025-03-14:
> If you remove all resistors, you should not have connection between the VRM pins A4, B3, B4 and A7, A8, A10, B7, B8, B9, B10.

i've checked this little guy and it is connected to A4, B3, B4, A7, A8, A10, B7, B8, B9, B10!
[image]

necroware, 2025-03-14:
Only that one? :) That's really unexpectedly small. But ok, if you remove it, will you still have connection between A4, B3, B4 and A7, A8, A10, B7, B8, B9, B10?

priangle, 2025-03-14:
i was too afraid to check the rest of them but now i see that they all lead to these pins :D the rabbit hole gets bigger and bigger

priangle, 2025-03-15:
> if you remove it, will you still have connection between A4, B3, B4 and A7, A8, A10, B7, B8, B9, B10?

nah, out of curiosity i removed this one res, and it didn't had any effect. 

another interesting finding is that after removing R38 and R27 there is still connection between their solder points

priangle, 2025-03-15:
look what i found on ebay - [mmx version of this board](https://www.ebay.com/itm/186911082217?_skw=ibm+socket+7+motherboard+mmx)

i think i will contact the seller and ask him to make a detailed top-down photo of VM area :))
[image]

priangle, 2025-03-15:
found another one, asked the same )
[image]

necroware, 2025-03-15:
> nah, out of curiosity i removed this one res, and it didn't had any effect.

If only one of those is left you will still have continuity through that, so either all or none :)
 
> another interesting finding is that after removing R38 and R27 there is still connection between their solder points

That is basically what I previously wrote. I'm also confused that it makes no difference if those resistors are installed or not. I think there are more of those and these resistors are there to distribute the current flow.

priangle, 2025-03-15:
another interesting finding
[image]

priangle, 2025-03-15:
[image]

necroware, 2025-03-15:
That looks like a different revision of this board. May be they realized, that there is a bug and remade the circuit?

priangle, 2025-03-15:
yeah, it's definitely a different revision, except that is the older one, dated 1994, mine is 1995.  
this looks like my revision, from 1995:
[image]

this is the older one:
[image]

main difference is the chips between memory and cache, and the sockets for memory upgrade for graphics chip

necroware, 2025-03-15:
> yeah, it's definitely a different revision, except that is the older one, dated 1994, mine is 1995. this looks like my revision, from 1995: 

I think, it doesn't on this image I can see a bigger SMD inductor, but on your board there is none.

[image]

priangle, 2025-03-15:
yeah, it is hidden underneath
[image]

necroware, 2025-03-15:
Yes, indeed. And this is an interesting picture. Can you see C92, C93, C112? C means capacitor, not resistor. And this is what I see on the other pictures, those are all capacitors. Most probably those  0 Ohm resistors connect I/O and VCORE voltage and have all to be removed and/or (optionally) replaced with capacitors.

priangle, 2025-03-15:
wow that is a very interesting find, there's indeed capacitors on c points on one of the photos of VRM-equipped board. i'll look into it, i think it would be right to remove resistors where it says C not R!

priangle, 2025-03-15:
well, i removed all 7 Rs around the VRM, and there's still connection between I/O and VCORE :D 
[image]

then i examined the rest of the board for Rs on Cs, and found one to the left of CPU. removed it and there's still connection. And then looked into the socket itself. O__O
[image]

So far it seems it would be easier to just hunt for the same board with VRM socket preinstalled. I ran out of ideas.

the only hope that some of ebay sellers i contacted will send me a detailed top down look on VRM area, but i doubt it will happen to be honest

necroware, 2025-03-15:
Those resistors inside of the socket are actually a good find too. Because the CPU planes are usually inside of the PCB and they are usually split exactly in the middle under the CPU socket. And all of those resistors are marked as capacitors too.

priangle, 2025-03-15:
so you think it's worth trying to remove them and see what happens? :)

necroware, 2025-03-15:
I mean you already got so far :)

priangle, 2025-03-15:
Wow. I removed all these Rs and now there's no connection between I/O and VCORE. Seems that it helped!
only between A4 and A7 there seems to be some Cap on the way (starts with some resistance, gets to "1" in about 2 secs (1 means no connection)), all other connections are no more!

Anyway it seems i need two detailed photos from the board with VRM socket - one of the VRM area, other from the CPU area. 

Maybe i should offer some money to the sellers for these photos :D

necroware, 2025-03-15:
Nice! I think all you would see from the pictures are the capacitors, which we already can see a little bit on the pictures which you've found. So I would just move on. Those caps which should be there instead of the resistors are optional and once you install the VRM you can measure the output voltage before you insert the CPU. The I/O voltage is not affected by the VRM module, it just wires it through or not, dependent how you set the jumpers. And the VCORE can be measured with a multimeter.

I guess it's time to build the VRM module ;)

priangle, 2025-03-16:
I have a few weeks before i receive my LCSC order, so in the meantime i might as well try to take a look at the mmx version of the board to make sure that everything's perfect:) not sure i can visually identify the parameters of needed caps though, but i think they're all identical around the board... 

btw, one of the sellers actually replied, and told he will ask his tech guy to make the photos i requested!

priangle, 2025-03-28:
Hey there! Huge update on my case!
since our last conversation i contacted 5 sellers, and none of them ended up sending the detailed photos of desired spots of the board. Yesterday i continued the search and tried every possible query and eventually stumbled upon the same board as mine equipped with VRM header - what's interesting, the board have a different part number and presented by the seller as salvaged from a completely different IBM machine. As a last straw of hope, i contacted him and voila! He answered almost instantaneously and today i received the cherished photos. 

Here are the side by side photos:

CPU socket area
[image]

VRM socket area
[image]

priangle, 2025-04-22:
Hey! The components arrived and i plan to build the module in a day or two. i was wondering, what's the safest way the check the module without damaging the CPU? You've mentioned that i can check the voltage without the CPU installed, can you elaborate please? 

Thanks in advance!

necroware, 2025-04-24:
Hi, sorry for the late answer. Great to see this getting there! To test this you can connect an external power supply to the +5V / GND pins on the VRM module and check if the VCORE output voltage corresponds to the setup on the switches. Just like I showed it in one of the videos about this VRM.

priangle, 2025-04-24:
Hey! Thanks for the reply! The module is finished, gonna try to test it and hopefully boot the pc later today!

[image]
[image]

necroware, 2025-04-24:
Nice! The coil looks a bit underrated, if the voltage gets unstable, or the coil too hot, try to get something beefier.

priangle, 2025-04-24:
> Nice! The coil looks a bit underrated, if the voltage gets unstable, or the coil too hot, try to get something beefier.

https://www.lcsc.com/product-detail/Power-Inductors_BOURNS-RLB0914-3R3ML_C1329807.html
this is the coil i've been recommended for this build, i'll look into how it performs as you suggested

priangle, 2025-04-25:
i tried to check the voltage of VCOREs without the CPU but i cant interpret the results. i connected the black needle to one of the Vss pins and the red one to VCOREs. here's the results:
PIN A7 - 5.045 v
PINS A8 B8 B9 - 2.246 v
PINS A10 B10 - 1.505 v
[image]

necroware, 2025-04-25:
I think you count the PINs from the wrong side. A1-2, B1-2 is ground, A14-15, B14-15 is +5V. You seem to count in reverse and if so, then the 2.246V you get should be VCORE. If you set 2.2V on the switches, then it looks right. The other voltages are strange, that should be 3.3V and ground if I count it right. Check out the pinout once again:

[image]

necroware, 2025-04-25:
3.3V for I/O comes from the mainboard and is not controlled by the S7-VRM. That voltage is specified to be between 3.3V and 3.5V so it should be fine.

priangle, 2025-04-26:
fully assembled the machine and everything seems to be in order. I think that means that the project is a success! as a precaution i just need to check the temp on the coil..
Thank you so much for your help and for this project project! I've renamed the issue