necroware/s7-vrm #10: No change with module

aderius, 2025-07-13:
Hello,

I put together the 0.6 version for the [Intel Advanced/ATX (Thor )](https://theretroweb.com/manufacturers/207)
Advanced/ATX (Thor). I changed the pin at J10K1 on the motherboard per the [manual](https://theretroweb.com/motherboard/manual/advanced-atx-4615-5f07978170bdd200457643.pdf) (and tried both settings for good measure). Regardless of how I configure the VRM, there is no change in Vcore voltage with or without it connected. I reassembled the VRM again on a new board just to be sure everything was soldered properly. I was wondering if you had any ideas regarding next steps.

necroware, 2025-07-13:
I have Thor on my TODO list. As far as I know it seems not be supported out of the box. It's not enough to solder the VRM header, also the board has to be modified as far as I saw. From the factory the I/O and Core voltage planes are not separated there and I don't know if it possible there. In the worst case Intel didn't follow its own specification on that board.

I have my a Thor at hand, but the mainboard is dead and needs to be repaired first. I take a look this week.

aderius, 2025-07-13:
Thank you very much for your reply! One thing I did try is removing the 0 ohm resistors on the board that connect to the vcore pins 22-25 (one of the IO pins has one as well) but that made no difference. Looking forward to what you find.

necroware, 2025-07-13:
You are lucky, I had some time today and this board was on my TODO list far too long. I repaired it and modified for the S7-VRM. You have to remove following 0 Ohm resistors, which connect I/O and Vcore planes from factory. If you don't remove all of those resistors, the on-board 3.3V/3.5V regulator will override the external VRM.

The 0 Ohm resistors to remove (sorry, the silk layer is somewhat hard to read, but I hope I got the names right):
- R2M5
- R3M4
- R3M2
- R3M5
- R3M6
- R4K3
- R5L1

[image]

aderius, 2025-07-14:
You are a legend - thank you very much!

aderius, 2025-07-16:
Belay that - when I insert a CPU, there seems to still be a connection between VIO and VCORE, thus circumventing the VRM. Without the CPU inserted, it's no problem. Looks like I have more investigating to do...

closed 2025-07-27 by necroware